esphome-econet/esphome-econet #468: Add "Emergency Heat" mode to climate control

jdanders, 2025-03-19:
I've got a dual-fuel furnace, and the thermostat supports "Emergency Heat" mode which is gas-only mode.
```
STATMODE : 5 (Emergency Heat)
```

I am not sure how to add it to ESPHome, as it's not one of the valid Climate modes. 

I recently added it to HA [core econet integration](https://github.com/home-assistant/core/pull/137623#pullrequestreview-2611668645). Within HA, they recommend adding a switch to represent Emergency Heat mode, so when in this mode the climate shows HEAT and the Emergency Heat switch is on. Normal heat is also in HEAT mode but with the switch off.

Another option (probably wrong) would be to map it to the `DRY` mode, because, ummm, gas is dry?

Any tips how it should be integrated in ESPHome climate?

jdanders, 2025-03-22:
I'm going to need some help, or else I'll just hijack `DRY` mode. Here's what I have working right now:
```yaml
climate:
  - platform: econet
...
    mode_datapoint: STATMODE
    modes:
      0: "HEAT"
      1: "COOL"
      2: "HEAT_COOL"
      3: "FAN_ONLY"
      4: "OFF"
      5: "DRY"  # EMERGENCY HEAT

...

switch:
  - platform: template
    id: emergency_heat_switch
    name: "Emergency Heat"
    lambda: |-
      return (id(econet_climate).mode == 5);
    turn_on_action:
      - climate.control:
          id: econet_climate
          mode: DRY  # EMERGENCY HEAT mode
    turn_off_action:
      - climate.control:
          id: econet_climate
          mode: HEAT  # HEAT mode
```

In order to have the switch tell the furnace to enter mode `5`, esphome limits the `mode` inputs to one of the predefined enumerations. What I would really like to do is to just push a 5 out to the furnace and have `climate` think that it's in `HEAT` mode. It would look like this:

```yaml
climate:
  - platform: econet
...
    mode_datapoint: STATMODE
    modes:
      0: "HEAT"
      1: "COOL"
      2: "HEAT_COOL"
      3: "FAN_ONLY"
      4: "OFF"
      5: "HEAT"  # EMERGENCY HEAT (now two modes report HEAT)

...

switch:
  - platform: template
    id: emergency_heat_switch
    name: "Emergency Heat"
    lambda: |-
      return (id(econet_climate).mode == 5);  #this will keep switch in sync with furnace status
    turn_on_action:
      - climate.control:
          id: econet_climate
          mode: 5  # EMERGENCY HEAT mode (this is illegal)
    turn_off_action:
      - climate.control:
          id: econet_climate
          mode: HEAT  # HEAT mode
```
There are two problems here. 
1. As mentioned above, `mode: 5` is not legal. I'm not sure how to send a 5 to the furnace when the switch is turned on.
2. I'm not sure what happens when two modes are called `HEAT`.

Any ideas on how to do this cleanly?

jdanders, 2025-03-24:
And you leave the `modes` as is? That results in some weirdness like this where it's in Cool Mode and EHeat at the same time:

[image]

The same sequence (Cool -> Emergency Heat transition) results in EHeat switch being on and mode showing Heat in econet cloud integration.

jdanders, 2025-03-24:
Yes, thermostat is fine, it's just HA that shows weird state. Personally I use EHeat mode every day to avoid peak-time power charges, so I need it automated in HA. 

The limitation in HA (and ESPHome) is why the cloud Econet integration uses an Emergency Heat switch. Switch on the switch and things go to EHeat mode, turn off the switch, they go to Heat mode. Turn on EHeat at the thermostat and the HA mode goes to Heat and the switch shows turned on.

tronikos, 2025-08-11:
If this ever gets implemented we should be careful to not cause #515