esphome-econet/esphome-econet #469: Outdoor and Supply temps are off by 100x on my furnace

jdanders, 2025-03-21:
See conversation: https://github.com/esphome-econet/esphome-econet/pull/467

Thermostat outside temp shows 44 while TEMP_OAT shows 43.7, so that's accurate. At the same time TEMP_EVP shows 35, which is likely accurate. Return air temp FURNRTMP shows 66, which is close enough to HVAC temp of 67.7 to be believable. 

The temperatures that don't work also don't change value: TEMP_CPT is -40F, outdoor_air_temperature and FURNSTMP show 6,514F.

jdanders, 2025-03-21:
I'm not sure what logs will be most helpful. This is the INFO level output of the device for about 13s.

[dump.log](https://github.com/user-attachments/files/19380102/dump.log)

justbruce, 2025-03-21:
> See conversation: [#467](https://github.com/esphome-econet/esphome-econet/pull/467)
>
> Thermostat outside temp shows 44 while TEMP_OAT shows 43.7, so that's accurate. At the same time TEMP_EVP shows [...]

The Crankcase (and base pan) heaters are optional on your model of ODU (RD17AZ48AJ3NA).

 If they aren't installed, I reckon the ODU simply reports the values as the minumum possible temp.

Do you know if your unit was optioned with a Crankcase heater at install?  If so, was it connected to the ODU Variable Speed Drive?

justbruce, 2025-03-21:
> I'm not sure what logs will be most helpful. This is the INFO level output of the device for about 13s.
> 
> [dump.log](https://github.com/user-attachments/files/19380102/dump.log)

This was helpful, as it shows your ODU reporting everything I'd expect it to without being specifically asked to  by either our little integration, the thermostat, or a technicians handheld gizmo.

It doesn't show your Furnace Model Number.  Would you mind providing that?  It might help with some troubleshooting

jdanders, 2025-03-21:
R6024-RD Coil: RCFZ6024STANMC (probably not relevant)
R96V85 Furnace: R962V0855A21M4C
RD748 ODU: RD17AZ48AJ3NA

I have no idea if I've got a crankcase heater.

justbruce, 2025-03-21:
> R6024-RD Coil: RCFZ6024STANMC (probably not relevant) R96V85 Furnace: R962V0855A21M4C RD748 ODU: RD17AZ48AJ3NA
> 
> I have no idea if I've got a crankcase heater.

Outdoor Air Temp and Supply Air Temp Sensors are "Optional field installed supply air and outdoor air.....(10K NTC thermistor)....shall be read from the T4 screw terminal block"  on your furnace according to it's manual

justbruce, 2025-03-21:
[image]

jdanders, 2025-03-21:
I have an OAT, as `TEMP_OAT` is reporting correctly. Are you wanting me to look and see if there's a supply sensor?

justbruce, 2025-03-21:
> I have an OAT, as `TEMP_OAT` is reporting correctly. Are you wanting me to look and see if there's a supply sensor?

Your ODU contains an Air Temp sensor.  The  TEMP_OAT Datapoint can be sourced [if I recall] from either the Furnace Sensor, the ODU Sensor, or both.

jdanders, 2025-03-22:
No sensors in the furnace, only ODU. 

[image]

jdanders, 2025-03-22:
In this case, would it be possible to detect the lack of sensor and report NaN?

justbruce, 2025-03-23:
> In this case, would it be possible to detect the lack of sensor and report NaN?

Theoretically, yes.  Seems wiser to simply disable this datapoint.  I've submitted a pull request to disable it by default ( #470  )