esphome-econet/esphome-econet #472: Very high update rate of many sensors, ignoring `econet_update_interval`

jdanders, 2025-03-25:
One of the variables exposed to override is `econet_update_interval` which I would want to set the update rate of all the sensors provided by this component. The `climate` module seems to respect it, but in my system everything else is providing updates much more frequently which drives up database size more than I'd like. Sometimes there are update periods as short as 1 second. Is it possible to rate limit all the sensors?

I don't know if this is expected behavior or my system is aggressively reporting status on its own. [This comment](https://github.com/esphome-econet/esphome-econet/issues/469#issuecomment-2742190167) by @justbruce made me wonder if normally status only comes when polled. It refers to this [dump.log](https://github.com/user-attachments/files/19380102/dump.log).

Sensors that I am sure are hyper-active:
Airhandler CFM
Airhandler RPM
Instant BTU Output
Instant BTU Usage
ODU Evaporator Temperature
ODU Expansion Valve Current Position
ODU Expansion Valve Super Heat (returns to 10 degrees between values)
ODU Inverter Crank Speed
ODU Lockout Timer
ODU Outside Air Temperature
ODU Pressure Suction (up and down from value to zero)
ODU Saturated Suction Line Temp (up and down from value to zero)
ODU Suction Line Temp
Static Pressure
Total Daily Gas

The comments in parentheses are about weird behavior besides high update rate. These are sensors that bounce up and down strangely when sampled. 

My proposed solution would be to add a `throttle_average` filter to most or all numeric sensors with the time period set to `econet_update_interval` but I'm not sure if there's a better way or some underlying issue that should be addressed.

jdanders, 2025-03-25:
Example sensor after filtering is applied:

[image]

justbruce, 2025-03-25:
> Sensors that I am sure are hyper-active: Airhandler CFM Airhandler RPM Instant BTU Output Instant BTU Usage ODU Evaporator Temperature ODU Expansion Valve Current Position ODU Expansion Valve Super Heat (returns to 10 degrees between values) ODU Inverter Crank Speed ODU Lockout Timer ODU Outside Air Temperature ODU Pressure Suction (up and down from value to zero) ODU Saturated Suction Line Temp (up and down from value to zero) ODU Suction Line Temp ODU Type (this is marked `request_once: true` but updates frequently?) Static Pressure Total Daily Gas

> My proposed solution would be to add a `throttle_average` filter to most or all numeric sensors with the time period set to `econet_update_interval` but I'm not sure if there's a better way or some underlying issue that should be addressed.

While I would agree there is something funny going on with the way sensors ar polled [in particular sensors which have request_once set true, but update multiple times per minute] I don't think averaging is the proper approach for most of the cases.

  HVAC is a unique situation where a majority of people ( I think)  want accurate, up-to-date numbers.  I want to know what the temperature is in the return duct/supply duct/ODU coil NOW, not a trend of the temperature over the past X number of samples.  

  I think tracking down the source of the erroneous updates and/or update rate is the proper fix, rather than papering over it with a filter.

jdanders, 2025-03-26:
Agreed that not all of these sensors should be averaged. Also agreed that there seem to be some other issues here than just high update rate that should be fixed.

In some cases, just `throttle` makes more sense than `throttle_average` (basically do I want the last sampled value, or do I want the average since last value?). ODU_Type is an example -- the average type doesn't make any sense. 

I recently made this change with my 32 power meter sensors. I wanted immediate data, but it was just too much history. Averaged throttle on the power numbers in that case made very little difference in the usefulness of the sensor, but wildly improved performance and cut my HA DB size in half.

The nice thing about both of those filters is if you want live data, you can lower `econet_update_interval`, which effectively disables the throttling. An update interval of 1s, for example, would be the same as if you didn't have a filter at all.

tronikos, 2025-03-27:
It's expected for some sensors to be updated more frequently than `econet_update_interval`, e.g. if you thermostat requests the datapoints.

For my HPWH to limit the state changes I have:

```yaml
sensor:
  - id: !remove ac_current_rms
  - id: !extend power
    filters:
      - round: 0
    accuracy_decimals: 0
  - id: !extend wifi_signal_sensor
    filters:
      - delta: 3.0
  - id: !extend ambient_temp
    filters:
      - lambda: return round(x/0.5) * 0.5;
      - delta: 0.5
  - id: !extend evaporator_temp
    filters:
      - lambda: return round(x/0.5) * 0.5;
      - delta: 0.5
  - id: !extend suction_temp
    filters:
      - lambda: return round(x/0.5) * 0.5;
      - delta: 0.5
  - id: !extend discharge_temp
    filters:
      - lambda: return round(x/0.5) * 0.5;
      - delta: 0.5
```

Since 0.5 accuracy for these diagnostic temperatures is more than enough. I need to see if your `throttle_average` filter would work better.

I also have the following automation to daily purge noisy sensors:

```
alias: Purge noisy sensors
description: ""
triggers:
  - at: "02:41:41"
    trigger: time
conditions: []
actions:
  - action: recorder.purge_entities
    data:
      keep_days: 1
      entity_globs:
        - sensor.memory_*
        - sensor.*free_memory
        - sensor.swap_*
        - sensor.processor_*
        - sensor.disk_*
        - sensor.*frequency
        - sensor.*voltage*
        - sensor.*current
        - sensor.current_1_*
        - sensor.*consumption
        - sensor.micro2wallbox*
        - sensor.solaredge_temp_sink
        - sensor.*_rssi
        - sensor.*_lqi
        - sensor.econet_hpwh_*
        - sensor.*_plug_power
        - sensor.thermal_zone*
        - sensor.iphone_*
        - sensor.*_electricalmeasurement
mode: single
```

jdanders, 2025-04-06:
I've noticed the `delta` filters all over the place. So the intent is to limit updates? 

An [example](https://github.com/esphome-econet/esphome-econet/blob/f9981a92dfba0f598044d3b822b89594188f5a0e/econet_hvac_furnace.yaml#L170C1-L173C1):
```yaml
    accuracy_decimals: 0
    filters:
      - delta: 0.9
```
I am confused by this, because I would think the `accuracy_decimals` setting effectively ignores the changes less than one unit. In other words, 63.4 and 63.3 would both be 63 and so no update would be needed. If so, then the `delta` filter is worthless. Maybe I'll do a quick experiment to see if esphome filters then update or just sends 63 twice.

Update on this issue: I decided to break out the throttle timing to provide flexibility if you want to get realtime data again. There will be a new variable:
```
  econet_update_interval: 30s
  econet_throttle_interval: ${econet_update_interval}
```
If you want to disable throttling, just set `econet_throttle_interval` to 1s. Maybe even 0s? I'll have to try...

tronikos, 2025-04-06:
You default `econet_throttle_interval` to `econet_update_interval`. I'm confused how this change will throttle updates. Won't updates still be sent every 30s by default?

From https://esphome.io/components/sensor/index.html#throttle-average

> For example a throttle_average: 60s will push out a value every 60 seconds, in case at least one sensor value is received within these 60 seconds.

As I said earlier your issue is likely your thermostat polls for data more frequently than every 30s.

Regarding `accuracy_decimals` and `delta` from https://esphome.io/components/sensor/index.html

> accuracy_decimals (Optional, int): Manually set the number of decimals to use when reporting values. This does not impact the actual value reported to Home Assistant, it just sets the number of decimals to use when displaying it.

> delta This filter stores the last value passed through this filter and only passes incoming values through if incoming value is sufficiently different from the previously passed one.

jdanders, 2025-04-06:
Thanks for explaining the `delta`, makes sense.

The intent is that a user can put this in their top level `yaml` to set the global update rate:
```yaml
substitutions:
  tx_pin: GPIO21
  rx_pin: GPIO20
  board: esp32-c3-devkitm-1
  platform: esp32
  variant: esp32c3
  econet_update_interval: 60s

packages:
  econet:
    url: https://github.com/esphome-econet/esphome-econet
    ref: main
    file: econet_hvac_furnace.yaml
...
```
This will result in both `update` and `throttle` intervals to 60s.

Or like this if they want to allow faster updates to sensors:
```yaml
...
  variant: esp32c3
  econet_update_interval: 60s
  econet_throttle_interval: 1s
...
```
This will set `update` to 60s and `throttle` to 1s.