esphome-econet/esphome-econet #492: Can't detect when serial connection is up or down

meyertime, 2025-06-26:
First of all, thanks for this project!  I have an Rheem electric water heater, and this has worked perfectly to integrate it with Home Assistant.  Seeing the readings from it even helped me fix it when the heating element failed.

Here's the issue that I'm having.  We are on an off-peak program with the electric company.  Basically, they shut off the water heater during certain peak times, and in exchange, we get over half off the cost of electricity.  When they do, we usually don't notice that the water heater is off until one of us tries to shower in the evening, even though it's usually shut off around 2pm.  I was hoping to set up some automation that would notify us when it's shut off and when it's back on.

The problem is there does not appear to be any way to detect when the water heater is shut off.  From the perspective of the ESPHome device, it would lose connectivity with the water heater.  In Home Assistant, all the readings stay the same and there are no events that I could trigger an automation on.

It's been a while since I set it up, so I'm on an oldish version of `esphome-econet`: `2.2.0` `2024.8.0`.  I checked the releases since then and didn't see anything that looks like it would address this.

How hard would it be to add a way to detect when the serial connection to the water heater is up or down?

meyertime, 2025-06-28:
Thanks for the tip!  I tried it...  It might work, but it's not perfect.  I set an automation to run when `last_updated` is greater than 10 minutes in the past, but it looks like `last_updated` only changes when the value of the sensor actually changes, not if it's updated to the same value, so I have to use a sensor that constantly changes in value.  The only sensors that change that frequently are the upper and lower temperature.  So far, there were some false triggers with the upper sensor.  I'll experiment to try to get it more reliable.

meyertime, 2025-06-30:
Thanks for the suggestion.  I don't think that will work in my case.  The ESPHome device is powered from a different circuit than the water heater, so it stays on when the water heater is shut off.  (This is a requirement, because off-peak devices must be hard-wired into the off-peak circuit, and only approved devices can be on that circuit.)  If my understanding of what you linked to is correct, I think it would still work without the connection to the water heater and could not be used to detect when the water heater is down.

tronikos, 2025-06-30:
My suggestion is to add the following in your yaml:

```yaml
sensor:
  - platform: econet
    name: "IRTC Seconds"
    id: irtc_seconds
    number_datapoint: IRTCSECS
    unit_of_measurement: "s"
    entity_category: "diagnostic"
    request_mod: 8
```

The IRTCSECS data point is polled from the water heater so if the water heater loses power you will stop getting seconds. The advantage over the temperature sensors is that this seconds sensor is guaranteed to change every `econet_update_interval` which defaults to 30 seconds.

tronikos, 2025-06-30:
I had a typo in my previous post which I've now corrected so make sure you use the code in the github post and not from your email notification.

meyertime, 2025-06-30:
Oh, ok.  Makes sense!  I tried that, and also updated to the latest version v2.8.1 (ESPHome 2025.6.2).  I'm seeing this output after connecting to the water heater:

```
[07:28:55][W][econet:349]:   IRTCSECS : UNSUPPORTED
[07:29:13][W][econet:349]:   HOTWATER : UNSUPPORTED
[07:29:13][W][econet:349]:   POWRWATT : UNSUPPORTED
```

Home Assistant shows `IRTC Seconds` now, but the value is `Unknown`.  It also shows `Hot Water` and `Power` as `Unknown`, but I think it was like that before.  Maybe my water heater does not support IRTCSECS?

meyertime, 2025-06-30:
Is there any way to query what data points are supported by my water heater?  Is there perhaps another one that might work for me?

tronikos, 2025-06-30:
There is no way that we know

tronikos, 2025-06-30:
Ideally we should change the state of all the sensors to unavailable when we lose connection with the water heater after some time, e.g. N times of econet_update_interval. Keeping this issue open.

tronikos, 2025-06-30:
The problem with last_successful_query or a counter is it increases the db size since every 30 seconds a new value will be transferred over WiFi and be persisted in the recorder database.
I intentionally optimized the component to avoid unnecessary updates to HA. In fact in my config I have additional sensor filters to further reduce updates.

meyertime, 2025-07-01:
Makes sense to me.  How hard would it be to change sensors to "Unavailable"?  Or at least one sensor (like heating element state)?  Or provide an up/down sensor?

meyertime, 2025-07-04:
Thanks for implementing this!  I see the new sensor now, but unfortunately, it's not working right. I cut power to the water heater for 10 minutes, and it still said "Connected". Then I physically unplugged the ESPHome device from the water heater diagnostic port, and within 2 minutes, it changed to "Disconnected". Then I plugged it back in (still no power), and it almost instantly changed to "Connected".

I'm not sure how it's detecting that it's connected to a device that has no power, but for my purposes, I need it to change when the water heater power is cut.

Let me know what I can do to help troubleshoot. I have OTA updates set up now, so I can try things out easily before you release.

tronikos, 2025-07-04:
Can you attach logs when you cut power to the water heater?

meyertime, 2025-07-04:
@barndawgie I don't see how it could stay online, because it has no power, and I can't imagine it has a battery backup.  I'm making an assumption, though.  What I do know is that the circuit feeding the water heater is shut off by the off-peak control relay, as if someone turned off the breaker.  So when I'm doing these tests, I'm turning the breaker off and on.

@tronikos Long story short, it appears that your `disconnected` branch fixed the issue.  I collected logs, though, and I am including them in case they're useful.  Here are the timings when reading the logs:

[v2.9.0.log](https://github.com/user-attachments/files/21068229/v2.9.0.log)
20:07:01 power cut
20:09:23 unplugged serial
20:11:45 plugged in serial
20:12:23 power restored

[tronikos-disconnected-branch.log](https://github.com/user-attachments/files/21068231/tronikos-disconnected-branch.log)
20:26:26 power cut
20:29:17 power restored

tronikos, 2025-07-05:
@meyertime if you want to reduce the time the sensor changes to disconnected you can adjust `mcu_connected_timeout: 120s`. Is 2 minutes too long for your use case? I just realized I didn't need to default it to such high value. We poll for each datapoint every 30s but because we make several different requests for different datapoints we end up making a request every 5s. Defaulting to 120s means the mcu failed to respond after 24 attempts. I think I'll reduce it to default to 30s.

meyertime, 2025-07-05:
It would take many minutes or hours for us to notice the hot water not being hot, so 2 minutes seems plenty soon enough to me.  As long as it's not so short as to get false disconnects.  Are there always several requests within 30s regardless of appliance type?

tronikos, 2025-07-05:
It now defaults to 30s. See PR description. Regardless of appliance type, the sensor will change to disconnected after at least 5 failed read requests.

PR #494 (2025-07-03): Add Microcontroller Connected sensor
Fixes #492