esphome-econet/esphome-econet #495: Multiple issues with Endeavor furnaces

slserpent, 2025-07-03:
I had a multitude of problems getting this project to work with my Endeavor line R962V gas furnace that I haven't yet been able to attribute to either the firmware/unit being newer (2023 first available) or there not being a communicating Econet thermostat (I thought this project could replace it, at least for diagnostics). If it is the latter, then perhaps the YAML configs should be separated so if one does not have the thermostat, the relevant datapoints and sensors can be omitted.

As to what isn't working, it appears most of the single result/setting change commands don't work, including AUTOMODE, STATMODE, F_ACTCFM, F_ACTRPM, COOLSETP, DHUMENAB, DHUMSETP, FAN_MODE, HEATSETP, HMTARGET, HUM_WFAN, HUMDCNFG, HUMDCTRL, HUMSMART, INDOORKW, RELH7005, RELHOFFS, SCHEDULE, SCRNLOCK, SPT, STAT_FAN, and STATNFAN. Obviously, many of these would never work without an Econet thermostat to provide the data like scheduling and setpoints. However, changing the fan speed or operating mode directly or getting fan speed seem like they should work without the thermostat.

The only single result commands that do return data are STATIC_P, FURNAFUE, HRHEEMID, and HWELL_ID. However, the last three all have issues. FURNAFUE returns 9000, which clearly doesn't line up with the expected 96% (or 4% waste). And HRHEEMID and HWELL_ID both have a different format than what is expected. The first string starts at 11 (instead of 4) and the second string starts at 35 (instead of 28). So there are 7 bytes extra that include some data that could be a possible header, maybe a protocol version number or some other identifier. Here's the raw data that HRHEEMID returns:

```
80 00 03 40 00 80 00 01 C0 00 3C 00 00 06 84 00
00 00 00 00 00 42 3C 00 00 52 39 36 32 56 31 30
30 35 41 32 31 4D 34 53 43 41 50 20 20 20 20 20
20 57 31 31 32 33 30 35 37 35 39 20 20 20 20 20
20 20 20 20 20 20 20 20 20 00 76 70
```

The only truly useful command to me is HWSTATUS, although again, the data has a lot of inconsistencies to what the project can currently decode. The most glaring is that the static pressure is only returned as FFFF. Very curious that the single result version STATIC_P works and yet the one from HWSTATUS does not.

I've come up with the following HWSTATUS data format through capturing and analyzing the raw return data in various operating modes, with comments on what data is and isn't working for me or what I'm unsure about. If someone with a similar model or firmware can confirm or compare values, that would certainly be useful.

```
// Econet response header
typedef struct {
    uint    : 1;
    uint   DestinationAddress : 31;
    ubyte  Unused;
    uint    : 1;
    uint   SourceAddress : 31;
    ubyte  Unused; //possibly part of next
    ubyte  DataLength;
    ubyte  Unused2[2]; //possibly part of next
    ubyte  Command;
} RX_HEADER;

// Econet HWSTATUS response
typedef struct {
    ubyte  MessageStart; // 84
    ubyte  Unused1[6];
    ushort Unknown1; // message type identifier?
    ubyte  Unused2[2];
    ubyte  heat_capacity; // not working? always 0
    ubyte  compressor_stage; // maybe not working unless stage 1 is given as 0
    ushort airhandler_cfm;
    ushort static_pressure; // divided by 5280. not working?
    ushort airhandler_rpm;
    ubyte  Unused3[2];
    ubyte  Unknown2; // Mode? 0 = Idle, 1 = Vent, 3 = Cool. 8 = Heat?
    ubyte  Unused4[3];
    ubyte  heat_stage; // not working? always 0
    ubyte  flame_sensor; // flame present?
    ubyte  Unknown3; // low pressure switch?
    ubyte  Unknown4; // high pressure switch?
    ubyte  Unused5[3];
    ubyte  Unknown5; // inducer speed? 2 = high, 1 = low, 0 = off?
    ubyte  Unused18[3];
    ubyte  FanOperating; // returns 1 whenever fan is operating, 0 otherwise 
    ubyte  Unused6[2];
    ushort Unknown6; // always 460
    ushort Unknown7; // always 2000
    ubyte  Unknown8; // always 2 - these three could be configuration values since they're static
    ubyte  Unused7[4];
    ushort Unknown9; // 17762
    ushort return_air_temperature; // divided by 10
    ushort outdoor_air_temperature; // divided by 10
    ubyte  Unknown10; // always 255
    ubyte  Unused8[47];
    ushort low_heat_2week_cycles;
    ushort high_heat_2week_cycles;
    ushort blower_2week_cycles;
    ushort powered_2week_cycles;
    ubyte  Unused9[1];
    ushort low_heat_lifetime_cycles;
    ubyte  Unused10[1];
    ushort high_heat_lifetime_cycles;
    ubyte  Unused11[1];
    ushort blower_lifetime_cycles;
    ubyte  Unused12[1];
    ushort powered_lifetime_cycles;
    ushort low_heat_2week_hrs;
    ushort high_heat_2week_hrs;
    ushort blower_2week_hrs;
    ubyte  Unused13[1];
    ushort low_heat_lifetime_hrs;
    ubyte  Unused14[1];
    ushort high_heat_lifetime_hrs;
    ubyte  Unused15[1];
    ushort blower_lifetime_hrs;
    ushort powered_lifetime_days;
    ubyte  Unused16[6];
    ushort supply_air_temperature; // divided by 10
    ubyte  Unused17[1];
} HWSTATUS_DATA;

// Econet CRC footer
typedef struct {
    ushort CRC;
} RX_FOOTER;
```

Finally, I also had some troubles on initial testing getting HWSTATUS to work at all. However it is setup by default results in a no polling taking place. I had to give it a `request_mod` group and give that group a `request_mod_update_intervals` definition for the polling to work. It certainly seems that it was intended that this command be polled, so I'm not sure how nobody else noticed it wasn't polling or alternatively why it wasn't polling for only me.

barndawgie, 2025-07-07:
> Finally, I also had some troubles on initial testing getting HWSTATUS to work at all. However it is setup by default results in a no polling taking place. I had to give it a `request_mod` group and [...]

Based on [a quick look at the conifg](https://github.com/esphome-econet/esphome-econet/blob/faac355da1c8352d7579a51dbb1d351cf69000bf/econet_hvac_furnace.yaml#L17),I think for HWSTATUS - and probably some other datapoints, especially for furnaces - we don't actively request them, but instead listen on the bus for requests and responses driven by a thermostat. That could explain your last point.

In general, I'm not sure if we have ever tested this with Endeavour furnaces - so there may be quite a few gaps. It seems like the CPP integration with the bus may work but that different ESPHome YAML configuration files are needed. If you can figure out a good config for one, we could figure out how to include it in the project.

If you need more detailed help, you might have better luck [on our Discord](https://discord.gg/cRpxtjkQQ3).

slserpent, 2025-07-07:
> Based on [a quick look at the conifg](https://github.com/esphome-econet/esphome-econet/blob/faac355da1c8352d7579a51dbb1d351cf69000bf/econet_hvac_furnace.yaml#L17),I think for HWSTATUS - and probably [...]

Ah, that does explain a lot. But in the case of not having the Econet thermostat, HWSTATUS definitely needs to be polled or this project is basically useless. I'm assuming a lot of these other commands are actively disabled without the thermostat as well, but I don't have a dump of the model's specific firmware to confirm whether these commands (datapoints?) are actually present or not.

> It seems like the CPP integration with the bus may work

Can you clarify what you mean here?

> different ESPHome YAML configuration files are needed. If you can figure out a good config for one, we could figure out how to include it in the project.

Well, I can share what works for me so far, but it's pretty rough. Really just waiting for someone with my HVAC model / without an Econet thermostat to help confirm the various aspects of what's different.

> If you need more detailed help, you might have better luck [on our Discord](https://discord.gg/cRpxtjkQQ3).

I did post there. Nothing much was said in response. That's Discord for ya.