esphome-econet/esphome-econet #515: Integration sometimes turns climate to HEAT on boot

jdanders, 2025-07-27:
I have noticed that when the ESP boots it will sometimes set the thermostat mode to HEAT. I've seen this happen from OFF to heat, and today it went from COOL to HEAT. 

I've seen it happen when I reconfigure/reprogram the device, and today it happened after a lost Wi-Fi for a minute.

I haven't done any investigation to root cause. Any ideas what might cause it?

barndawgie, 2025-07-27:
Can you share a little more about your scenario? What type of unit do you have? Are you using basically out "stock" YAML or you have some customizations? What versions of stuff are you on?

tronikos, 2025-07-27:
My bet is on a Home Assistant automation that incorrectly triggers from unavailable state. In the history you should be able to see which one it was.

jdanders, 2025-07-28:
Thanks for the quick responses. I've got some changes, see my [fork](https://github.com/jdanders/esphome-econet).  I don't see any automations causing this.

Here's a log of the bad behavior. I uploaded a new image, and when the system came up it was in HEAT mode. Home Assistant logging doesn't show any automations.

https://pastecode.io/s/1bpqnpob

Here's the HVAC logbook:

[image]

jdanders, 2025-07-28:
Hmm, that log seems incomplete, as it doesn't include a STATMODE response.

I've done it a couple more times, and it seems to be happening every time I restart the ESP. I originally installed things during the winter, so I probably didn't notice it because HEAT was what I wanted. Here's a better log:
https://pastecode.io/s/7oxxagu2

This is the first clear indicator I see of the mode, but it's status from the furnace, not a command sent from the ESP:
```
[10:08:10][I][econet:395]: Read 225. ms=8877
[10:08:10][I][econet:162]: <<< 80.00.03.40.00.80.00.03.80.00.9E.00.00.06.13.82.00.00.00.0E.48.65.61.74.69.6E.67.20.20.20.20.20.20.20.07.80.00.00.42.82.00.00.07.80.00.00.42.90.00.00.07.80.00.00.00.00.00.00.07.80.00.00.40.00.00.00.0B.82.00.00.00.06.4F.66.66.20.20.20.0B.82.00.00.01.06.4C.6F.77.20.20.20.0F.82.00.00.02.0A.4C.6F.77.20.43.6F.6F.6C.20.00.0C.82.00.00.01.07.43.6F.6F.6C.69.6E.67.08.82.00.00.00.03.4E.6F.20.07.80.00.00.42.90.00.00.08.82.00.00.00.03.4F.66.66.0E.82.00.00.00.09.4F.66.66.20.20.20.20.20.20.0B.82.00.00.00.06.4F.66.66.20.20.20.77.F0 (174)
[10:08:10][I][econet:164]:   Dst Adr : 0x340
[10:08:10][I][econet:165]:   Src Adr : 0x380
[10:08:10][I][econet:166]:   Command : 6
[10:08:10][I][econet:167]:   Data    : 13.82.00.00.00.0E.48.65.61.74.69.6E.67.20.20.20.20.20.20.20.07.80.00.00.42.82.00.00.07.80.00.00.42.90.00.00.07.80.00.00.00.00.00.00.07.80.00.00.40.00.00.00.0B.82.00.00.00.06.4F.66.66.20.20.20.0B.82.00.00.01.06.4C.6F.77.20.20.20.0F.82.00.00.02.0A.4C.6F.77.20.43.6F.6F.6C.20.00.0C.82.00.00.01.07.43.6F.6F.6C.69.6E.67.08.82.00.00.00.03.4E.6F.20.07.80.00.00.42.90.00.00.08.82.00.00.00.03.4F.66.66.0E.82.00.00.00.09.4F.66.66.20.20.20.20.20.20.0B.82.00.00.00.06.4F.66.66.20.20.20 (158)
[10:08:10][I][econet:339]:   STATMODE : 0 (Heating)
[10:08:10][I][econet:310]:   HEATSETP : 65.000000
[10:08:10][I][econet:310]:   COOLSETP : 72.000000
[10:08:10][I][econet:310]:   TEMPOFFS : 0.000000
[10:08:10][I][econet:310]:   DEADBAND : 2.000000
[10:08:10][I][econet:339]:   STAT_FAN : 0 (Off)
[10:08:10][I][econet:339]:   STATNFAN : 1 (Low)
[10:08:10][I][econet:339]:   HVACMODE : 2 (Low Cool)
[10:08:10][I][econet:339]:   AUTOMODE : 1 (Cooling)
[10:08:10][I][econet:339]:   AWAYMODE : 0 (No)
[10:08:10][I][econet:310]:   SPT_STAT : 72.000000
[10:08:10][I][econet:339]:   HW_G_FAN : 0 (Off)
[10:08:10][I][econet:339]:   FURNGFAN : 0 (Off)
[10:08:10][I][econet:339]:   VACA_FAN : 0 (Off)
```

I sat by the thermostat for one of my tests and the thermostat changed from COOL to HEAT immediately once the ESP rebooted.

I would guess that it's related to the fact that HEAT mode is `0` and ESP is setting mode before getting the update from the furnace. I'll keep investigating...